lushaylabs/tangnano9k-series-examples #2: uart example not working tang Nano 9K

mhanuel26, 2023-07-05:
HI @lushaylabs ,

I am testing the uart example on the Tang Nano 9K , when I program the devices all led goes oOn but using terminal serial port does not seems to change any of them. I am selecting the only COM port that ios connected when I turn on the board, so COM pot should not be the problem, I have tested using the  VSC Serial Console with same result. What others things can I try to check?

Thank you in advance for your comments,

lushaylabs, 2023-07-08:
Hey, to first make sure it is not a synthesis issue, could you try the .fs file that is in the repo [here](https://github.com/lushaylabs/tangnano9k-series-examples/blob/master/uart/uart.fs) just place this in the repo instead of your own fs file (call it the same name as your current .fs file) and then program it using the "Program Only" option from the "FPGA Toolchain" button or using openFPGALoader directly.

This .fs file has been tested so it will let us know if it is a synthesis issue which might require different compilation flags or versions, or if it is com port issue, where you might need a different driver / port in-order to interact.

Let me know if it works or not and we can take it from there

mhanuel26, 2023-07-13:
hI @lushaylabs ,

I can confirm the original .fs file does the job. I haven't modified the code, so it is a tool synthesis issue?
What would you suggest to address this?

This is the output console when build and debug

```
Processing uart.lushay.json
Starting FPGA Toolchain
Starting Yosys CST Checking
     /----------------------------------------------------------------------------\
     |                                                                            |
     |  yosys -- Yosys Open SYnthesis Suite                                       |
     |                                                                            |
     |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
     |                                                                            |
     |  Permission to use, copy, modify, and/or distribute this software for any  |
     |  purpose with or without fee is hereby granted, provided that the above    |
     |  copyright notice and this permission notice appear in all copies.         |
     |                                                                            |
     |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
     |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
     |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
     |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
     |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
     |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
     |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
     |                                                                            |
     \----------------------------------------------------------------------------/
    Parsing uart.v
        - Module uart parsed
    Checking if all ports are defined in constraints file
    All Ports are defined
Finished CST Checking
Starting Synthesys with Yosys (Gowin)
    Step 1:     Executing Verilog-2005 frontend: c:\Dev\tangnano9k-series-examples\uart\uart.v
    Step 2:     Executing SYNTH_GOWIN pass.
        Step 2.1:   Executing Verilog-2005 frontend: c:\OSS-CA~2\bin\../share/yosys/gowin/cells_sim.v
        Step 2.2:   Executing HIERARCHY pass (managing design hierarchy).
        Step 2.3:   Executing PROC pass (convert processes to netlists).
        Step 2.4:   Executing FLATTEN pass (flatten design).
        Step 2.5:   Executing TRIBUF pass.
        Step 2.6:   Executing DEMINOUT pass (demote inout ports to input or output).
        Step 2.7:   Executing SYNTH pass.
        Step 2.8:   Executing MEMORY_LIBMAP pass (mapping memories to cells).
        Step 2.9:   Executing TECHMAP pass (map to technology primitives).
        Step 2.10:  Executing OPT pass (performing simple optimizations).
        Step 2.11:  Executing MEMORY_MAP pass (converting memories to logic and flip-flops).
        Step 2.12:  Executing OPT pass (performing simple optimizations).
        Step 2.13:  Executing TECHMAP pass (map to technology primitives).
        Step 2.14:  Executing OPT pass (performing simple optimizations).
        Step 2.15:  Executing IOPADMAP pass (mapping inputs/outputs to IO-PAD cells).
        Step 2.16:  Executing OPT_CLEAN pass (remove unused cells and wires).
        Step 2.17:  Executing DFFLEGALIZE pass (convert FFs to types supported by the target).
        Step 2.18:  Executing TECHMAP pass (map to technology primitives).
        Step 2.19:  Executing OPT_EXPR pass (perform const folding).
        Step 2.20:  Executing SIMPLEMAP pass (map simple cells to gate primitives).
        Step 2.21:  Executing ABC pass (technology mapping using ABC).
        Step 2.22:  Executing TECHMAP pass (map to technology primitives).
        Step 2.23:  Executing OPT_LUT_INS pass (discard unused LUT inputs).
        Step 2.24:  Executing SETUNDEF pass (replace undef values with defined constants).
        Step 2.25:  Executing HILOMAP pass (mapping to constant drivers).
        Step 2.26:  Executing AUTONAME pass.
        Step 2.27:  Executing HIERARCHY pass (managing design hierarchy).
        Step 2.28:  Printing statistics.
        Step 2.29:  Executing CHECK pass (checking for obvious problems).
        Step 2.30:  Executing JSON backend.

    Summary
[... 107 lines cut ...]
    ConfDataLength: 712
    ProgramDoneBypass: OFF
    SPIAddr: 00fff000
    SecurityBit: ON
    idcode: 1100481b
    loading_rate: 0
    Jtag frequency : requested 2.50MHz    Done
    erase Flash Done
    Flash Written
    Done
    Error: Error key checkSum not found
    CRC check: Success
    displayReadReg 0001f020
    	Memory Erase
    	Gowin VLD
    	Done Final
    	Security Final
    	Ready
    	POR
Finished FPGA Programming
Toolchain Completed
```

Thank you,
Manuel

lushaylabs, 2023-07-15:
Sorry for the delay, took some time to test, it seems to be an issue due to a change done in the opensource toolchain with your version of OSS CAD Suite. I know I told you to download that version, so apologies for the confusion. I have retested all the projects using this version https://github.com/YosysHQ/oss-cad-suite-build/releases/tag/2023-02-10 and have not had any issues. Sorry again for the inconvenience and thank you for reporting the issue. 
The open source toolchain is currently going through a redesign and then I will update the recommended version from `2023-02-10` to the latest version once tested

Krasnomakov, 2024-04-27:
Hello, 

I am following the UART tutorial. When it comes to Tabby testing or interacting with Node via serial it is denying the access both in Tabby and when use a script. Why? How can I allow the access? 

node .\serial-program.js
node:internal/process/promises:289
            triggerUncaughtException(err, true /* fromPromise */);
            ^

Error: Opening COM9: Access denied
Emitted 'error' event on SerialPort instance at:
    at SerialPort._error (C:\Users\user\Downloads\lushayLabs\tangnano9k-series-examples\uart\node_modules\@serialport\stream\dist\index.js:85:18)
    at C:\Users\user\Downloads\lushayLabs\tangnano9k-series-examples\uart\node_modules\@serialport\stream\dist\index.js:118:18

Node.js v20.12.2

lushaylabs, 2024-04-29:
@Krasnomakov Im have had trouble with the UART on windows before, not sure if it is the same issue as you are having but from what I saw it is usually a driver issue.

I had to move `usbser.sys` from `C:/Windows/system32/drivers/` (I moved it to my desktop instead) then I had to uninstall the driver in device manager for the comm port device. I then removed the device and replugged it in and instead of the previous driver it used a lib-usb driver. I uninstalled this as-well and installed the sipeed driver from here https://dl.sipeed.com/shareURL/TANG/programmer

I also downloaded the latest version of ZADIG to replace the driver for interface 0 to winusb

After that I have interface 0 using winusb and interface 1 (the serial port) using ftdibus

I am including some images of how the drivers look. Let me know if yours looks similar
[image]

[image]

Krasnomakov, 2024-04-29:
Thank you, issue resolved completely. Tested in Tabby, Serial Terminal and with the script.

Krasnomakov, 2024-04-29:
@lushaylabs Thank you, issue resolved completely. Tested with Serial Terminal, Tabby and the script.