lushaylabs/tangnano9k-series-examples #5: is there any PLL module? how I can get freq over my quartz?

fufellen, 2023-11-13:


lushaylabs, 2024-04-29:
Hey @fufellen, nextpnr-himbaechel should support PLLs you can test it out using our preview version of OSS-Cad-suite with this https://github.com/lushaylabs/oss-cad-suite-build/releases/tag/2024-04-21

and you can find example code in apicula here https://github.com/YosysHQ/apicula/tree/master/examples/himbaechel/pll

This is still relatively new so it might be a bit complicated to use still, we are working on adding a PLL generator to our VSCode extension

axsmyth, 2024-12-13:
Hi. I appreciate your code very much.
Sorry for offtopic, but your site lusshaylabs.com appears to get out of DNS now. Please delete this issue if it's wrong.

lushaylabs, 2024-12-13:
Thank you for bringing this to my attention, it seems like an issue with my hosting provider, looking into it

axsmyth, 2024-12-22:
Hi! One more small issue. Some pictures at learn.lushaylabs.com/led-panel-hub75 are unreachable to load.

axsmyth, 2025-01-01:
> Hi! One more small issue. Some pictures at learn.lushaylabs.com/led-panel-hub75 are unreachable to load.

Oops it happens again. Subdomen learn.lushaylabs.com does not recognize by DNS or whatsoever. There is "Domain error
Failed to resolve DNS path for this host" only.

lushaylabs, 2025-01-02:
Hey @axsmyth thanks again for the report I have been trying to get it back up to no success yet, 

The site can still be accessed directly via https://lushay-labs.ghost.io/ but for some reason the global DNS is not working, hopefully will have it back up soon