lushaylabs/tangnano9k-series-examples #7: Flash Memory example only goes to first 16k sector

SoulCalDan, 2024-04-15:
Using the lushaylabs example below. I believe the flash.lushay.json file to configure the external flash on the Tang Nano 9k only goes to the first 16kB. I have created code to load a 383kB bmp file, but after the fist 16kB, values are all FF (the default for the flash chip) so my image is only partially loaded. Is there a way to expand the .json or the openFPGAloader to write to end of file so all bytes are pushed to the flash?

https://learn.lushaylabs.com/tang-nano-9k-reading-the-external-flash/

My .json file loads a 383kB bmp

{
    "name": "flash",
    "externalFlashFiles": ["test.bmp"]
}

lushaylabs, 2024-04-29:
Hey @SoulCalDan sorry for the delay, is your project public I would like to debug it on my end to try and get to the bottom of it

SoulCalDan, 2024-04-29:
I shouldn't have to provide any project files. The json script from the lushaylabs page just stops writing to flash before any entire file is written. The cutoff for that seems to be 16kB. The example only has a handful of characters to display for the OLED screen, so larger amounts of data probably weren't considered for the script.

lushaylabs, 2024-04-29:
After testing it I think it is a synthesis issue and not a flash issue. Like the flash module is not correctly functioning after a certain address. I recompiled it using the new nextpnr himbaechel and it seems to work as expected, I loaded a 120kb file and was able to view It all 

I am attaching a version of oss-cad-suite that has nextpnr-himbaechel which you can try https://github.com/lushaylabs/oss-cad-suite-build/releases/tag/2024-04-21

I am also attaching a link to the .fs file which you can try and program your tang nano with to verify that the issue is with the synthesis

 https://drive.google.com/file/d/15P8fNrxFJuP39AhtLW37Ss5n6Jgz9dLv/view?usp=sharing

SoulCalDan, 2024-05-14:
I tried the flash.fs file to load into two separate Tang Nano 9K with Visual Studio Code, and the result was the bytes in flash all are FF after 0x3AF8. I updated with oos-cad-suite-windows-x64-20240421.exe, and now the flash goes to 0x4B00 before it all is FF.
[image]
[image]

lushaylabs, 2024-05-14:
I am sorry to hear that, I have tested this on my end and I was able to get past that area, so if you are using the same fs file then we can rule out synthesis issues
[image]

To manually flash an external file using oss-cad-suite you can double-click on the `start.bat` file inside the oss-cad-suite folder which will open up a terminal with the open source toolchain loaded into the terminal's path

You can then run:
```
openFPGALoader.exe -b tangnano9k --external-flash  <file name>
```

As in:
```
openFPGALoader.exe -b tangnano9k --external-flash C:\Users\lushay\Downloads\test.bin
```

You can also add `--verify` before the file name to make it verify the contents of the flash after writing

SoulCalDan, 2024-05-15:
Manually running the start.bat seemed to work. However, it took a much longer time to run than expected, around 8 minutes for just a 383kB file on this small 4MB of memory. Using the Visual Studio Code json file like before seems to make the write to flash memory much quicker, but obviously that didn't work (maybe it is returning a false done?)
Below is my output from the start.bat and running the openFPGAloader.exe command.

[OSS CAD Suite] openFPGAloader.exe -b tangnano9k --external-flash --verify tiny.bmp
empty
write to flash
←[94mJtag frequency : requested 6.00MHz    -> real 6.00MHz   ←[0m
←[94mParse file ←[0m←[32mDONE←[0m
after program flash: displayReadReg 0003f020
        Memory Erase
        Gowin VLD
        Done Final
        Security Final
        Ready
        POR
        FLASH lock
←[94mErase SRAM ←[0m←[32mDONE←[0m
←[33mJtag probe limited to %d MHz6000000←[0m
←[94mJtag frequency : requested 10.00MHz   -> real 6.00MHz   ←[0m
Detail:
Jedec ID          : 85
memory type       : 60
memory capacity   : 16
Detail:
Jedec ID          : 85
memory type       : 60
memory capacity   : 16
RDSR : 00
WIP  : 0
WEL  : 0
BP   : 0
TB   : 0
SRWD : 0
←[33mflash chip unknown: use basic protection detection←[0m
Erasing: [←[0m==================================================←[94m] 100.00%←[0m←[32m
Done←[0m
Writing: [←[0m==================================================←[94m] 100.00%←[0m←[32m
Done←[0m
←[94mVerifying write (May take time)←[0m
Read flash : [←[0m==================================================←[94m] 100.00%←[0m←[32m
Done←[0m