lushaylabs/tangnano9k-series-examples #10: UART Testbench WaveTrace problem

Greydesk, 2024-05-31:
Hi there,
Just got my TangNano9k yesterday and I'm working through the Lushay tutorials. I am using the software recommended EXCEPT I'm using the 20240531 version of oss-cad-suite. Everything seems to run fine except when I go to look at the produced VCD file. I get the error:
The editor could not be opened due to an unexpected error: Assertion Failed: argument is undefined or null
I'm guessing this is from the oss-cad-suite version but I'm just checking to make sure.
Thanks!

Greydesk, 2024-05-31:
Update. I have confirmed that using 20230210 produces a viewable vcd file.
However, I'm getting a CRC Error when trying to program the device with this code. Weird.
    Error: CRC check : FAIL
    Error: Read: 0x0000431b checksum: 0xb4bb
    Error:

Greydesk, 2024-05-31:
Maybe its the way I'm using VSCode. I notice that the second lesson doesn't have a 'top' module. Am I setting this up the wrong way?

lushaylabs, 2024-05-31:
Hey, 

Not sure about the waveform issue, was it on a specific example ? I will try and reproduce it with the version you mentioned. Wavetrace doesn't use oss-cad-suite so if changing the version fixed it, it must be something inside the generated vcd file which caused it to crash

Regarding the CRC issue I was able to reproduce, it seems from a specific version of oss-cad-suite this problem happens, I will look into it and try and track down the exact tool causing the issue to try and solve this issue. From my experience I was still able to program the chip, but just received this error in addition, is that what has happened to you ?

And regarding the UART example you are right, the "top" module in that example is not called `top` the module named `uart` is the module. "top" is the default name that works out of the box, and you can change the top module either in the project file like here https://github.com/lushaylabs/tangnano9k-series-examples/blob/master/uart/uart.lushay.json or by passing the `-top` flag to `synth_gowin` if synthesizing manually

Greydesk, 2024-06-05:
So, several of my problems seem to have been caused by how I was using VSCode. I had created a directory for my programs and was using that for my workspace. This is wrong. You need a different workspace for each program/tutorial. 

Once I created a new workspace and created the files again (I copied and pasted from the old files), the testbench ran fine and the output worked with vcd viewer. This is with the 20240531 version of oss-cad-suite.
I created the json file you linked and compiled the program and uploaded it and it worked great. 
On a secondary run through, it gives the same CRC fail. If I load the first tutorial and build and program, it also loads, gives the CRC error, but functions. Going back to the second tutorial, build and program, it also loads, gives the CRC error, but functions. I think that there is either something odd with the oss-cad-suite or a problem with my nano9k.