lushaylabs/tangnano9k-series-examples #14: Problem with UART - no tty device detected

fedor-goncharov, 2024-08-11:
Hi, I did the tutorial on UART on my Tang Nano 9k, flashed with no errors but I struggle to communicate with 
board via UART: 

```OS : ubuntu 23.10```

my dmesg on plug-in
```
[28637.616311] usb 1-2: New USB device found, idVendor=0403, idProduct=6010, bcdDevice= 5.00
[28637.616320] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[28637.616323] usb 1-2: Product: JTAG Debugger
[28637.616325] usb 1-2: Manufacturer: SIPEED
[28637.616327] usb 1-2: SerialNumber: FactoryAIOT Pro```

lsusb 
```Bus 001 Device 008: ID 0403:6010 Future Technology Devices International, Ltd FT2232C/D/H Dual UART/FIFO IC```

For fun ```openFPGALoader --detect```

```No cable or board specified: using direct ft2232 interface
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
index 0:
	idcode 0x100481b
	manufacturer Gowin
	family GW1N
	model  GW1N(R)-9C
	irlength 8```

fedor-goncharov, 2024-08-11:
To add : I use recommended version of OSS tools, native Gowin Programmer tool detects two "cables" (as far as I understand, "cables" correspond to devices in Linux) -> one cable allows to do all memory resets when I need (I guess it
is a JTAG interface) and the other throws "Cable open failed"

closed 2024-08-15 by fedor-goncharov

lushaylabs, 2024-08-15:
Hey @fedor-goncharov thank you for reaching out and sorry for the slow reply,  just to check did you try testing the fs file from [here](https://github.com/lushaylabs/tangnano9k-series-examples/blob/master/counter/counter.fs) as-is, it uses an older version of OSS-CAD-Suite before the CRC issue so it would be interesting to see if it works.

I saw you were having some trouble erasing the flash all of a sudden on the tang nano 9k, if you can send me an email (contact@lushaylabs.com) I will try debug it with you there

I also haven't tested Ubuntu 23, so I will give that a try over the weekend and update the docs accordingly once we get this working.